so-artn/ARTN #1: Autofocus routine

bjweiner, 2017-10-13:
We need a routine that will take a series of frames stepping through the focus, and then an analysis script that evaluates the images (eg with astropy photutils?) to find the best focus value.

RTS2 should be able to change the focus with "focpos=2300", "focpos+=50" etc, but when Sam and I tried this, it did not work as expected.

srswinde, 2018-01-03:
I played with the focus script a little bit last night. Focus moved as expected, but the analysis didn't go well. To make it work we need to:

1.  make sure the source extraction software (SEP) is really getting correct sources
2. find out what happens when there are very few sources, 
3. should we flatfield and bias subtract first (probably). 

There are also the competing focus routines, Mike Lessers, CSS' Those might be better directions.

dsand, 2018-01-24:
Petr, please provide Scott with the 'focus analysis script' that can take the pause+shift images and determine the best focus.

srswinde, 2019-09-15:
We went with T.E. Pickering's script and it works pretty well--closing this one.

closed 2019-09-15 by srswinde