so-artn/ARTN #5: Autologging exposures

bjweiner, 2017-10-13:
As images are taken, write a human-readable log summarizing useful fields: image name, target name, filter, exptime, UT, RA, Dec, airmass. Ideally have some kind of autologger window for a user to write comments.

dsand, 2017-11-14:
The CfA instruments for the MMT/Magellan have this kind of autologger...is this what you were thinking of, Ben? Should we ping Brian McLeod, or someone similar, to get a copy of their code? I worry it is written in Tcl/Tk though.

bjweiner, 2017-11-20:
Yes, I was thinking of something like that. We could ask Brian, but theirs might be too hardwired to their data format etc. I can write a script pretty easily that goes through a directory and extracts useful numbers from the FITS headers. But it would be nice to have it be live and allow comments.

Another useful thing that the CfA does for Hectospec is to save a skycam image when the data is taken, so you can go back and see what conditions were like. We should be able to do this with the Bigelow skycam.

srswinde, 2017-12-18:
Information about each observation is kept in the rts2 sql database (stars). You can display this information using the rts2-nightreport command. If we want allow users to add to this interactively I suggest a web app.

srswinde, 2017-12-18:
I like the idea of an all sky camera image for each observation. I can start logging the allsky images and we can figure how to display this in whatever form the observation app takes later.

srswinde, 2017-12-23:
I wrote a program in the gui/db directory called nightobs that interacts with RTS2 sql observations table. From here we can pull up individual observations and match them with the exposures taken and provide a way for observers to comment on the quality of observations.

closed 2020-09-14 by dsand