so-artn/ARTN #6: Moon Angle Constraints for Targets

dsand, 2017-11-03:
Is it possible to add moon angle constraints for input targets? A default value should be something like >25 deg, but maybe this can be user specified as well.

bjweiner, 2017-11-03:
rts2-target has an argument "--lunarDistance lower:upper". I don't know the units but possibly degrees?  Blank values for lower and upper indicate no limit. So for example adding "--lunarDistance 20: --airmass :2.2" to the rts2-target call in Sam's read_superlotis.py script may do something that we want.

bjweiner, 2017-11-06:
I tried this out and it appeared to work - it rejected two targets that were <15 deg from moon. Scott had to give the rts2obs account write access to the /etc/rts2/targets directory for it to go through. That has an overall constraints.xml file and then a separate directory for each target with its own constraints.xml.  The xml file is pretty easy to read and hand-edit if you need to. We could also write a script that generates an rts2-target command line call.

I had a separate problem where it rejected some southern targets as below the horizon when they weren't. I don't understand that one yet.

srswinde, 2017-12-18:
I think this is a done deal.

closed 2017-12-18 by srswinde

bjweiner, 2017-12-18:
Great, yes, I was just typing here: I think my previous comment was an issue with the horizon constraints that has been fixed (not sure of the details). The "--lunarDistance lower:upper" syntax works, so we just have to keep using it.