so-artn/ARTN #7: Easy creation of observing scripts

bjweiner, 2017-12-01:
We need an interface that will allow users to easily specify an observation without them needing to know the RTS2 script format.

I made a step in this direction with the script make_newobs.py, which will read a simple text catalog that specifies 1 target per line with ra, dec, desired filters, exposures and repeats. It can also be run to type the numbers in interactively. They still need to give numbers for the filters rather than names.

To do: figure out how to do a dither pattern. and to translate filter names to numbers. If the script can read a file of filter wheel names/numbers somewhere on the observing computer, that would help.

srswinde, 2017-12-19:
I added the ability to read the filter positions from RTS2. I need to incorporate this into the rest of the make_newobs.py stuff

srswinde, 2019-09-15:
This will be taken care of by ORP and [rts2solib](https://github.com/srswinde/rts2solib)

closed 2019-09-15 by srswinde