so-artn/ARTN #9: Next level scheduling test -- Observe an object on a 3 hour cadence

dsand, 2017-12-18:
We would like to push on the scheduling abilities of rts2 -- Dave Sand will look into observing objects with a cadence tonight.  One example would be 'Observe this object for 2x60s in V and 2x60s in R roughly every 3 hours'.

pkubanek, 2017-12-19:
tempdisable {seconds} in observation script + targets in circular queue should do that. See man rts2.script for help. The problem is that queues probably did not check for temporary restrictions. So the only way would be to modify start time associated with the target in the queue and put it at back of the queue. I am pretty sure we considered such observations and queue shall do it, the question is how. I will keep looking.

dsand, 2018-01-28:
Here is an appropriate target for this type of scheduling.  Science case: Finding RR Lyrae in Milky Way dwarf galaxies.

RA: 10h34m52.8s DEC: +51d55m12s

B: 2x150s  V: 2x150s R: 2x150s

Twice per night, roughly 3 hours apart.

closed 2020-09-14 by dsand