so-artn/ARTN #11: Fits template file doesn't seem to work

srswinde, 2017-12-19:
According to rts2.template man page, you can create a fits template file and populate the fits headers with various rts2 values, basically anything in the rts2-mon left hand column can be put in the fits header. I tried this and it didn't seem to work. 

Here is what I did:

- created a file called /etc/rts2/fitstemplate.ini with the lines:
`[PRIMARY]`
`FOCUS = @F0.FOC_POS / Focus postion provided by RTS2`

- added the following line to the rts2.ini file:
`template = "/etc/rts2/fitstemplate.ini`
- restarted rts2 and took a dark

FOCUS was not added to the fits header.

bjweiner, 2018-01-24:
Petr says location of fitstemplate file needs to be specified on startup - rts2.ini, I think.

dsand, 2018-04-28:
Talking to Scott tonight, this has been resolved and was fixed by Petr.  As of April 27 2018, it is in /etc/rts2/fitstemplate.ini

closed 2019-11-04 by dsand