so-artn/ARTN #18: Flat fielding script did not did not execute

dsand, 2018-02-03:
[I'm copying an email I sent to Petr + Scott here]
1st email:
Hey guys --
RTS2 and the telescope started up like a dream tonight.  It even went to try to do a set of flats (/etc/rts2/flat.py), but it is stuck, and not taking any data.  Attached is a screenshot I took with my phone.  You can zoom in and see what is going on.   (see img_1311)

2nd email:
It sat there for about 10 min, and now it seems to be looking for targets.  See new picture. (see img_1312)

[image]
[image]

dsand, 2018-03-31:
Just FYI.  This still seems to be an issue.  Tonight the RTS2 took biases, and then we think it tried flats, but we could not find the images.  Petr, can you look at tonight's log or observations to figure this out?

dsand, 2018-04-28:
On April 27, we attempted to run flats at the beginning of the night.  As the appropriate time approached, in rts2-mon on the camera driver we received a warning that the priority had changed, and then the flats did not execute.  By restarting RTS2, we were able to get the flats to start.

srswinde, 2018-05-17:
Flats have worked the last two ARTN nights. Its not clear why

bjweiner, 2020-01-15:
From the ARTN meeting today, Scott says the flatfielding script needs cleaning up. One issue is that it doesn't fail gracefully if it tries to take a flat in a filter that is not installed, meaning lines have to be commented out if a filter is changed (and then they might not get changed back if the filter is re-installed).

dsand, 2020-01-21:
We have done some updating of this script tonight.  Added back U-band and Open filters.  Changed script to flat_artn.py.

closed 2020-09-14 by dsand