so-artn/ARTN #19: Write Script that calculates focus from focus-sequence images

dsand, 2018-03-02:


srswinde, 2018-03-02:
Last night I added an rts2 python script that initiates the focus shifting. You can find it [here](https://github.com/bjweiner/ARTN/blob/master/scripts/shiftfocusing.py). Eventually we will put the focus calculator in here as well.

you can test this script by associating it with a target in rts2 like:
```bash
rts2-target -c C0 -s "exe /home/rts2obs/ARTN/scripts/shiftfocusing.py" <targetid>
```
or by running it with rts2-scriptexec:
```bash
rts2-scriptexec -d C0 -s "exe /home/rts2obs/ARTN/scripts/shiftfocusing.py"
```
and the shiftfocus will be done wherever the telescope is in the sky currently. 

Right now the file is written to shiftfoc_<currentdate>_.fits, which isn't great because it will overwrite each time. I will try to fix this.

bjweiner, 2018-05-05:
Tonight rts2 changed the focus away from the value of 2300 we had determined by hand, winding up around 2480. I think it may be trying to run an autofocusing routine, and I don't know how to enable/disable that. I didn't catch this for a long time and suspect a lot of the data is out of focus.

srswinde, 2018-05-05:
To turn off focus run set the shiftfocus value in rts2-mon C0 to false. Though you should never have to do this, it should be done automatically.

It may have become confused because we changed it with the xephem gui.

 Another possibility is that I am confused about the focuser settings on RTS2. There are 5, they are written below with their RTS2 descriptions:

- FOC_POS  -> focuser postion
- FOC_TAR  -> focus target position
- FOC_DEF -> default target value
- FOC_FOFF -> offset from focusing routine
- FOC_TOFF -> Temprorary offset for focusing

I think I move the FOCUS_TAR when I do the focusing script. Maybe this is wrong.

dsand, 2018-08-27:
Quick update.  I think we want a 1) a good starting value and 2) a good map for seeing how the focus changes with telescope position and temperature.  

I think we have this information now.  

1. Based on Betsy's email here: [Betsy email](https://github.com/bjweiner/ARTN/tree/master/focus_related/Green_info)
We can set the initial focus to 2400, or something close, before our initial focus run?

2. Go to:[CSS_info](https://github.com/bjweiner/ARTN/tree/master/focus_related/CSS_info) 
This shou.d have the CSS-derived focus map on the sky, and info on how the focus changes with temp.

2.

srswinde, 2019-09-15:
We are now using T.E. Pickerings focus analysis script which works if there are enough stars on the CCD (5-10). We should come up with some focus fields that will assure we get enough stars at whatever exposure time we deem necessary.

closed 2020-09-14 by dsand