so-artn/ARTN #33: Kuiper did not move to requested target

srswinde, 2019-04-19:
BIG61 moved to 14:43:15.230 +056:03:24.30 requested 14:42:42.020 +056:08:10.24 target 09:39:10.670 +032:14:12.10 

This happened for SN2019bie on night of 4/17/2019

srswinde, 2019-04-19:
BIG61 moved to 16:31:31.180 +046:32:57.30 requested 16:30:58.000 +046:35:18.20 target 09:39:10.670 +032:14:12.10

This as well.

bjweiner, 2019-05-01:
This is the issue where RTS2 seemed to start an exposure before the telescope had settled, and then we were on the wrong field. For the first field, SN2019bie, the correct coordinates are 14:42:38 +56:08:10. The above is a cut and paste from the rts2-mon log: 
BIG61 moved to 14:43:15.230 +056:03:24.30 [wrong coordinates] requested 14:42:42.020 +056:08:10.24 [right coordinates] target 09:39:10.670 +032:14:12.10 [I don't know what these numbers represent]

bjweiner, 2019-05-01:
I believe the second object where this occurred was SN2019cvz at 16:30:54, +46:35:18. The log line from rts2-mon was 
BIG61 moved to 16:31:31.180 +046:32:57.30  [wrong coordinates]
requested 16:30:58.000 +046:35:18.20  [correct coordinates]
target 09:39:10.670 +032:14:12.10
In both cases the "requested" coordinates are 4 sec of time later than the actual SN coordinates. We think this is the ~1 arcmin bump that is applied to the coordinates to keep the object away from the bad column & amp division, so the "requested' are where we wanted to go.

closed 2020-09-14 by dsand