so-artn/ARTN #35: CRVAL1 and CRVAL2 are now being set to 0

tepickering, 2019-09-19:
The correct RA and Dec are being populated in the RA and DEC headers, but are not being populated into the WCS headers CRVAL1 and CRVAL2.

pkubanek, 2019-09-19:
This is most likely easy to fix. Just add CRVAL[12] to FITS template with
the correct variable.

srswinde, 2019-09-19:
I believe this is working now for at least the primary header. We will have to restart RTS2 to see if it is working for the 2 extensions. 

Thanks Petr!

dsand, 2019-09-19:
Hey Petr -- We are still having problems with this.  We can populate the primary header, but we cannot seem to populate the image extension headers.  Even when we put [CHANNEL1], [CHANNEL2] in the init file.  Any chance you can take a look and see what we are doing wrong?

pkubanek, 2019-09-19:
Sorry, got to it only now. You need to provide EXTNAME in [], CHANNEL[12] is just example (from LSST CCD testing days). I edited fitstemplate to:

```ini
[IM1]
CRVAL1 = @BIG61.TEL:ra
CRVAL2 = @BIG61.TEL:dec
SCTEST = TEST / this is a test

[IM2]
CRVAL1 = @BIG61.TEL:ra
CRVAL2 = @BIG61.TEL:dec

SCTEST = @W0.filter
SCTEST2 = FOOBAR / do you see this on chanel
```

give it a try (kill -1 $(pidof rts2-executor) shall be enough, if not, restart rts2-executor.

srswinde, 2019-09-19:
I will implement the IM1 and IM2 stuff soon.

dsand, 2019-11-04:
I believe this is working now--Closing

closed 2019-11-04 by dsand