so-artn/ARTN #39: Slewing to DEC=+60:11:34

pmilne511, 2019-09-19:
SN 2017eaw is at DEC=+60:11:34.9
One plan attempted to slew to that position, but ended up at +58:30:11.0
The sequence of 8 images ran with the TCS monitor showing "DEC" and "S1" 
blinking yellow (presumably because it didn't get to +60. The plan ran again 
(probably a mix-up loading plans), and the second time the telescope moved to the 
correct position. This probably just points out a grey zone between the various 
limits, but as this showed...it can happen that we are right at the limits where RTS2 
accepts the target, but the telescope doesn't quite get there.

dsand, 2019-09-19:
> Indeed now that we have a new and safer limit regime in place, I think we
> should limit targets to above the first set of software limits. In Dec that
> is 58 degrees. I like telling people that if you want to get above 58
> degrees you can't Use ARTN to do it.
>
> Is this too restrictive?

NGC6946 is at +60, but has had ~10 SNe.  SN2017eaw among them.  It would be
nice to squeeze this into our limits.
Other important galaxies are also at high declination.  M81, M82.

Aside from my desires on DEC limits, what do we need to do to ensure that
if someone requests an inappropriate target (e.g. at +75 DEC) that it does
not get attempted?  Do we need to clean up the ambiguity between
hardware and software limits so that the situation discussed here
(SN2017eaw) does not get attempted either?

srswinde, 2019-09-19:
In my above comment, I was talking about the multi tiered [TCS limits](https://lavinia.as.arizona.edu/~tscopewiki/doku.php?id=tcs:limit_logic).  There are also limits in RTS2 that I can change. I believe they are currently set to +60 in Declination.  RTS2 won't try to observe anything outside its limit the targets will languish in the queue forever--So no worries there. 

The idea with the +58 software Dec limit is that a user can go through it but they have to hit the goto button again when they reach it, ensuring they are watching what is happening. It would be silly to have RTS2 do this. 

That means our choices are to bring the TCS S1 software limits up to RTS2 at +60 or bring the RTS2 Limits to +58. It sounds like Dave's preference is to set it all to +60. 

I could

pmilne511, 2019-09-19:
RTS2 did accept +60:11  last night..............................the
telescope just took two slew calls to go there.
Looking at the images, I know it did go to the SN.

dsand, 2019-09-19:
I agree. It appears something is wrong here if RTS2 tried to move to a +60
DEC object.

closed 2020-09-14 by dsand