so-artn/ARTN #40: Suggestions for AORP page

pmilne511, 2019-10-01:
Monday, Sept 30th was a night full of potential host galaxies of a GW event. This meant that there were 90+ targets in addition to SN/LBV targets. Trying to manage this queue in combination with the non-infrequent system failures led to a hectic night, with subpar efficiency. The multiple kernel panics were a pain, but in this note I mention ideas to improve the utility of the AORP. 
1) When a target is sent to the queue (the eye icon), the page returns to default settings. This means that if I have navigated to a page with 3 targets I want to load.....the webpage jumps back to page one/default sorting after each selection. When 90 targets had to be selected, with many on page 9 or 11 or 12.....this led to a LOT of additional clicking. It would be good to preserve the sorting and page number after selecting a target. There can be a default (or reset) button to get back to default. 
2) When one uses the upper "next" button to page to the next page, going from page 1 -> 2 leads to a shift....which puts the mouse right over the "logout" button. It makes it too easy to accidentally logout when having to repeatedly (50 times) click from page 1 -> 9 . It would be good to move the logout. button to not be above the "next" button, or to prevent the page jumping when next is clicked. 
3) It would be wonderful for the list of records to show if the target is currently in the queue, maybe also showing if it has been observed this session. With a very long list, it is rough to have to keep switching tabs to the web-app page to see if an individual target has been loaded already. Remember that there are 13 pages of targets now and 100 targets were to be loaded tonight. Perhaps there can be a column with a letter or icon showing whether the target is already loaded and/or has been observed.

tepickering, 2019-10-01:
holy carp! even trying to manage a tenth as many targets between the AORP and observing queue was an enormous hassle. can't even imagine dealing with this...

i give a hearty 👍 to each of @pmilne511's suggestions. in the world of GW event follow-ups, this kind of load should be expected to be the norm and not the exception. there needs to be better operational interaction between the AORP and the observing queue. the current method is way too error-prone. we also need better tools for batch submissions to the AORP as well as batch methods to select from AORP and submit to queue.

bjweiner, 2019-10-01:
I agree that the best way to handle this is a sort of batch load, where an entire set of targets is treated as one package to send to RTS2. For GW search for MMT/Binospec, the MMT and SAO people implemented a batch mode where the user uploads a list of targets, and then the entire pile is sent to the instrument without the observer having to individually select them. It's possible to make the execution very fast or very slow by choosing a good or bad order of targets to minimize slews. For RTS2, typically RTS2 chooses the order of targets (like WESTEAST), which is at least ok, but one could imagine trying to optimize that further.

dsand, 2019-10-01:
You're a hero, Peter!

pndaly, 2019-10-01:
Will take a look at all this when I'm back in the office on 13 Oct. All good suggestions. I think some
are either done or in progress but, perhaps, not advertized as being available. Thanks.

pkubanek, 2019-10-02:
Hi Petr & al,

sitting on AstroRob, I just heard about gwtool developed by Watcher
folks.

https://gwtool.watchertelescope.ie/

It should be relatively easy to run something similar (code is open
source) on UofA servers and sends the list to Kuiper (and any other
queue enabled telescopes).

Petr

dsand, 2019-10-02:
Hey Petr --
We in fact already use this. [Its a little buggy]
Dave

pkubanek, 2019-10-02:
Well I mean we can write Python scrip which will:

- search/create target galaxies, either using the site or in house version

- pick it me visible (west-east sorted)

- observe the fields, report back images

As we will know which one were observed, script (another) can revisit the
fields later to take second epoch for substraction

Looks too trivial to be made - I might start something if nobody will kick
it on.

Petr

dsand, 2019-10-02:
OK, lets talk about it first.  Like I said, we are already doing this at
some level.

pmilne511, 2019-10-02:
How about meeting at Scott's office at 10? I am leaving home, so Scott
please text the location.

dsand, 2019-10-02:
Peter
I sent an email to the whole group yesterday saying we are meeting in 550.
I booked the room and others are meeting us there!

srswinde, 2019-10-02:
Let's do it at my office. Steward Annex. I will unlock the front door.

Sent from my phone. Please forgive my brevity and typos.

closed 2020-09-14 by dsand