so-artn/ARTN #46: Get github repos in order

dsand, 2020-06-01:
This is almost two issues. 

1. Right now there are at least two github repos that have ARTN-related material in them: mo-ops and this one, ARTN.  Once could imagine the need for a third -- a Mont4k pipeline.  Should we consolidate these two or three?  Or have multiple repos, but very clear definitions for each. Lets continue discussions, at least for a little while.

2. There are several active repos, at least for mo-ops, and it is not always clear which is the 'active' one at the telescope.  It is also true that the main branch code on the ARTN repo does not always seem to be in use at the telescope.  We should have clear 'development' and 'in use' repos, or something similar.  And stay up to date at the telescope.

dsand, 2020-09-14:
Progress has been made on this with the new so-artn github project:
https://github.com/so-artn
still work in progress.  Lets keep going on github cleanup.

danavner, 2021-02-03:
Everything in github matches what is on at the telescope.

closed 2021-02-03 by danavner