so-artn/ARTN #49: Retire Issues prior to June 2020?

dsand, 2020-09-14:
Wonder if we should retire all of our issues prior to June 2020, or archive them?  Just to clean up our issue list a bit and refocus our efforts on the tasks at hand.  

Thoughts, @danavner?  I wonder if there is a way to archive issues.

tepickering, 2020-09-14:
the way i've seen this handled in the past is to create an `archived` label, tag the issues to archive with that, and then close them. that way you won't see them by default, but you can easily search for them by the label.

dsand, 2020-09-14:
I like this.  I'll work on some more issues I need to generate to give @danavner some time to contribute.  And then I might just do it.

danavner, 2020-09-14:
I like that idea as well. Let's move forward with it.

dsand, 2020-09-14:
Done.  And closing this issue.  Victory.

closed 2020-09-14 by dsand