so-artn/ARTN #50: AzCam + RTS2 interaction issue coming out of 2020 shutdown

dsand, 2020-09-14:
To quote from a Dan Avner email to Lesser:

We ran into a serious issue with how AzCam/RTS2 interact. After taking an image, rts2 would wait indefinitely for a response from AzCam saying that the image is done. The telescope would not slew to the next target or continue to take the next image. 

Our next run isn't until the end of October, but I can head up any time before to work with you resolving the issue. Either we need to change the azcam driver on our end or if you made changes on feedback from azcam, revert back to the previous feedback.

dsand, 2020-09-14:
Dan and Lesser are actively working on this.

danavner, 2020-09-14:
We are currently working on rolling back some summer shutdown changes for azcam by making the mode generic. AzCam expects commands to come in as "rts2.reset" or "rts2.set_roi" however the current rts2 driver calls commands as "reset" or "set_roi". I asked Mike to rollback the config pre shutdown. 

In the next coming weeks we will test the system.

srswinde, 2020-09-14:
[https://github.com/srswinde/rts2/blob/master/src/camd/azcam3.cpp#L647](here) is an example of where you would change that in the RTS2 driver.

danavner, 2021-02-03:
This was fixed during our next run. Mike L. rolled back the changes he made from summer shutdown to the more generic azcam format.

closed 2021-02-03 by danavner