so-artn/ARTN #56: JS9 Image Seeker

dsand, 2020-09-15:
- Test getting image with @pndaly 's seek code
        - remote-obs-website/project/views.py
        - function get_images
        - change the directory parameter to point to the correct place for images.  Might need to make a form input on the js9 dashboard tab for user input directory or make the get_images smart enough to know what directory to automatically point at.
        - Phil's seek code is in remote-obs-website/project/utils
        - To test, go to the 10.30.1.24/js9/images/1. Grab the last hour of images.

danavner, 2021-02-03:
I  still need to fix the directory issue. WIll be working on it soon.

danavner, 2021-02-03:
Heheh meant to comment not close with comment.

danavner, 2021-09-16:
Sam wrote code to do this. Finished and closing.

closed 2021-09-16 by danavner